linorobot/linorobot2 #150: Firmware installation

mervinn11, 2025-04-06:
Recently when I tried to upload the firmware it shows successfully uploaded but not working, like the Teensy Orange light is blinking which is not supposed to happen

hippo5329, 2025-04-08:
It might be imu issues. Check the i2c wirings and pull up.

There is a test_sensors utility in my hippo5329/linorobot2_hardware. Check the wiki.

If this is your first micro-ros project, you should start with the re-connection example.

hippo5329, 2025-04-08:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#micro_ros_arduino_examples_platformio

hippo5329, 2025-04-08:
https://github.com/hippo5329/linorobot2_hardware/wiki#connect-the-imu-and-test_sensors

Or, comment out the imu in config. Then try bringup to verify the micro_ros connection.

PaulBouchier, 2025-06-08:
@mervinn11 please close this issue and re-file in linorobot2_hardware - it belongs there. Thanks!

closed 2025-07-27 by PaulBouchier