linorobot/linorobot2 #152: Docker Documentation

atticusrussell, 2025-04-19:
Hi maintainers,

Thank you for all your hard work on this excellent project. I see that there's a `docker` directory that contains docker-compose, Dockerfile, entrypoint, configuration settings, etc. 

I was wondering if whoever created it (@grassjelly)? might be able to create a `docs/docker.md` or a `docker/README.md`, that outlines the intended usage of the setup - whether it's intended to be used for development, deployment, etc. I would love it to be clear for myself how to best leverage it.

Thank you!

hippo5329, 2025-04-21:
Please check the wiki,

https://github.com/hippo5329/linorobot2_hardware/wiki#docker

PaulBouchier, 2025-07-27:
Closing as answered

closed 2025-07-27 by PaulBouchier