linorobot/linorobot2 #168: Jazzy branch gazebo still broken with turtlebot3_world

PaulBouchier, 2025-06-23:
Thanks for the quick fix to the initial location of the robot @grassjelly . However there are still multiple problems causing it to not be able to navigate "out of the box" using the instructions in the README.md:

1. The default map loaded by ros2 launch linorobot2_navigation navigation.launch.py is still set to playground. This is in linorobot2/linorobot2_navigation/launch/navigation.launch.py.
2. After providing an override for the map with map:=turtlebot3_world.yaml, I see that whereas the world instantiated by gazebo has a vertex of the hexagon directly ahead of the center of the world, the map file has a flat side of the hexagon directly ahead. IOW the map orientation doesn't match the orientation of the gazebo world.
3. The robot won't navigate when given a goal. The window running navigation.launch.py prints, 
[component_container_isolated-1] [WARN] [1750642777.672551818] [collision_monitor]: [scan]: Latest source and current collision monitor node timestamps differ on 1750642666.972544 seconds. Ignoring the source.
continuously. This is a bit different than the usual message when the robot is too close to an obstacle, and I'm not sure what's causing it.

It should be possible (according to the README) to run the following commands in three terminal sessions:
ros2 launch linorobot2_gazebo gazebo.launch.py
ros2 launch linorobot2_navigation navigation.launch.py
ros2 launch linorobot2_viz navigation.launch.py
and the defaults should make it just work, and the user should be able to set the initial pose on the default map and then navigate wherever. they want in the world That doesn't work.

PaulBouchier, 2025-06-26:
Hey @grassjelly I'm sorry to tell you there are still multiple problems causing gazebo to be broken. I hate to see this because you worked so hard to make harmonic work on Jazzy. Docker is particularly problematic. You referred to docker so I took a closer look at it. It looks like this will take a while to work through, and I wonder if you ought to revert the jazzy branch to before the turtlebot3_world change, which worked. But there are problems even there - see below. Changing the world to turtlebot3-world broke so many dependencies.

I pulled and tested the 166-jazzy... branch and here are my findings:

**Navigation issues**
1. The fix to the default map in navigation.launch.py works - all good there
2. The map of the turtlebot3_world is still wrong. The launch file provides the path to the correct map in linorobot2_navigation/maps/turtlebot3_world.yaml. Look at the two images below: the gazebo world has a vertex beyond the cylinder in front of the robot, but the map has a flat side beyond the cylinder in front of the robot

[image]

[image]

I suspect that the map is rotated 90 degrees relative to the gazebo world. You can see in the second image that the lidar scan does not match the map. So No. 2 is not fixed.
3. It still won't navigate, so No. 3 is not fixed. See below an image of the path to a goal, but the navigation window still prints continuously:
`[component_container_isolated-1] [WARN] [1750913070.239776569] [collision_monitor]: [scan]: Latest source and current collision monitor node timestamps differ on 1750912407.839767 seconds. Ignoring the source.
`
I checked out all the way back to the PR-159 merge and I still get this error, so I suspect a recent navigation2 update broke something. Do you update your ROS debs regularly? (apt update && apt upgrade)? I don't know why you wouldn't be seeing this issue.

**Docker issues**
1. I had written documentation in the README.md on how to use docker - see Quickstart section 1.1c. The instructions don't work anymore because of multiple changes in the docker compose and Dockerfile files. Could you please update the README.md as appropriate. I see there's a new sim.yaml which looks like it should launch multiple panes but I don't know how to invoke it. The previous docker compose target navigate-sim doesn't exist anymore.
2. I deleted the previously built image and ran docker compose build per the instructions in Installation section 4. It failed to build. (It did build before I deleted the image, the problem is early in the Dockerfile. It's possible there is a typo is the locale installation. The Dockerfile says on line 14: 
`&& locale-gen en_US.UTF-8 \`
but the Jazzy install instructions say 
`sudo locale-gen en_US en_US.UTF-8`
IOW the Dockerfile is missing en_US after locale-gen.
I get the following errors in build:
`
 => ERROR [base ros2 2/5] RUN apt-get update && apt-get install -y         locales         && locale-gen en_US  221.6s
------                                                                                                                 
 > [base ros2 2/5] RUN apt-get update && apt-get install -y         locales         && locale-gen en_US en_US.UTF-8         && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8     && rm -rf /var/lib/apt/lists/*:                          
18.79 Ign:1 http://packages.ros.org/ros2/ubuntu noble InRelease                                                        
18.79 Ign:2 http://archive.ubuntu.com/ubuntu noble InRelease                                                           
18.79 Ign:3 http://security.ubuntu.com/ubuntu noble-security InRelease                                                 
37.22 Ign:1 http://packages.ros.org/ros2/ubuntu noble InRelease
37.22 Ign:4 http://archive.ubuntu.com/ubuntu noble-updates InRelease
37.22 Ign:3 http://security.ubuntu.com/ubuntu noble-security InRelease
55.65 Ign:1 http://packages.ros.org/ros2/ubuntu noble InRelease
55.65 Ign:5 http://archive.ubuntu.com/ubuntu noble-backports InRelease
55.65 Ign:3 http://security.ubuntu.com/ubuntu noble-security InRelease
74.09 Ign:2 http://archive.ubuntu.com/ubuntu noble InRelease
77.16 Err:1 http://packages.ros.org/ros2/ubuntu noble InRelease
77.16   Temporary failure resolving 'packages.ros.org'
77.16 Err:3 http://security.ubuntu.com/ubuntu noble-security InRelease
77.16   Temporary failure resolving 'security.ubuntu.com'
....
147.9 E: Package 'locales' has no installation candidate
`

I checked out versions all the way back to gz-harmonic and can't get it to build from scratch. Are you able to build from scratch? I wonder if some dependency changed, because I previously built it and tested it to write the README instructions. Very puzzling.
3. Docker launches the robot in the middle of the center cylinder. This is because docker-compose.yaml spawns the robot at offset 0,0 in the gazebo target, which is in the middle of the center cylinder
4. The Dockerfile has duplicated sections: install language and install timezone are duplicated

I think the first priority should be to get what used to work, working again (the PR-159 version, and the docker compose build). I will look at these, but welcome your help. A second priority would be to get as updated as we can and have everything still work. I don't know what motivated the change of worlds to turtlebot3_world but it broke lots of stuff and a full test is needed to requalify the codebase.

Sorry to bring bad news.

PaulBouchier, 2025-06-26:
I'm puzzled by some of the steps in the Dockerfile. I hadn't looked at the docker build carefully before, but now I see some things that make me wonder:

1. It builds from the official ros dockerhub repository, and pulls ros:jazzy-ros-base, so why is it doing all the base stuff in the ROS install instructions like locale, timezone. timezone is installed by jazzy-ros-core so doesn't need to be re-installed. ros-jazzy-ros-base also sets up LC_ALL and LANG, so installing locales shouldn't be needed. It also installs some of the base packages like gnupg2, cur. The linorobot2 Dockerfile could be simplified by not building everything.
2. In the past, I pulled from the osrf/ros: collection, which has these images: https://hub.docker.com/r/osrf/ros/tags. Some of the images in there are interesting, like jazzy-simulation and jazzy-desktop-full-noble, so it seems if the linorobot2 Dockerfile pulled from there, it wouldn't have to install gazebo harmonic.
3. The .env causes cyclonedds to be installed and used. fastrtps is the default for ROS. Why use cyclonedds? There might be advantages to using the default. 

IOW it looks like there could be simpler and more stable ways to build the linorobot2 docker image. Are you open to changes?

PaulBouchier, 2025-06-26:
Please ignore the previous noted issue about not building in docker. I have removed it from the posts above. This was a problem with my system config, in which I just enabled ufw. Because docker was already installed, it seems the right rules weren't in place to allow apt in a container to reach http destinations. Uninstalling and re-installing docker solved the problem.

PaulBouchier, 2025-06-26:
Thanks @hippo5329 - it's the docker compose file and its .env file that sets cyclonedds. AFICT the code running natively would use fastrtps. So micro-ros agent is not a concern here. I just didn't understand why it is set that way.

hippo5329, 2025-06-26:
We need micro-ros agent for this project. I had removed the cyclonedds setting in my dockefile. You might found many people said cyclonedds worked better than fastrtsp in earlier ROS2 distros. The Moveit project also recommended cyclonedds. But micro-ros services and actions never work with cyclonedds. In my earlier tests, even basic micro-ros  msg did not work with cyclonedds on Jazzy. So it is good to remove cyclonedds on Jazzy and later.

grassjelly, 2025-06-26:
thanks Paul for testing. In summary, for the initialization related issue, we're only left with `turtlebot3_world and its map are oriented differently` ?. Regarding this, wouldn't this be fixed by setting the initial pose to the correct orientation?

PaulBouchier, 2025-06-27:
Sounds good - thanks @grassjelly

closed 2025-06-27 by grassjelly